nfc-tools/libnfc #629: Don't write the entire tag

TheRaf974, 2020-11-16:
I'm running Kali with libnfc 1.8.0 and I have a ACR122U and a clear Mifare Classic 1k

When I use the write command: `nfc-mfclassic W a u new-tag.mfd old-tag.mfd`
I have this output:
`NFC reader: ACS / ACR122U PICC Interface opened
Found MIFARE Classic card:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04  
       UID (NFCID1): 13  d6  9f  62  
      SAK (SEL_RES): 88  
RATS support: no
Guessing size: seems to be a 1024-byte card
Writing 64 blocks |............................................................|
Done, 60 of 64 blocks written.`

But when I check the data in the tag with `mfoc -f mifare-classic-keys.txt -O test-reading`, I have this:
`13 D6 9F 62 38 88 04 00 C8 44 00 20 00 00 00 15 
01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 
01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 
FF 00 FF 00 FF 00 FF 07 80 00 FF 00 FF 00 FF 00 
01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF 
01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF 
01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF 
...`
instead of :
`13 D6 9F 62 38 88 04 00 C8 44 00 20 00 00 00 15 
01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 
01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 
FF 00 FF 00 FF 00 FF 07 80 69 FF 00 FF 00 FF 00 
01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 
01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 
01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 
FF 00 FF 00 FF 00 FF 07 80 69 FF 00 FF 00 FF 00 
01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 
01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 
01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 
FF 00 FF 00 FF 00 FF 07 80 69 FF 00 FF 00 FF 00 
01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 
01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 
01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 
FF 00 FF 00 FF 00 FF 07 80 69 FF 00 FF 00 FF 00
...`

The soft only wrote the first sector and the first block of each other sectors.

However, when I write the tag with MifareClassicTool on Android, it works, except for the UID but it's normal because my tag is a first generation.

Can you please help me guys?

---------------------------------------------------------------------------------------------------------------------------------------------------------------------
Distributor ID: Kali
Description:    Kali GNU/Linux Rolling
Release:        2020.4
Codename:       kali-rolling
---------------------------------------------------------------------------------------------------------------------------------------------------------------------
ACR122U
P/N: ACR122U-A9
S/N: R-012156

msschulte, 2021-01-01:
Unfortunately, I have the same problem.

In the same context:
[https://github.com/nfc-tools/libnfc/issues/629](url)

msschulte, 2021-01-01:
Unfortunately, I have the same problem.

In the same context: [#632]

thekix, 2021-01-01:
Hi,

I am testing it with my ACR readers and I don't have this problem.

- Could you test libnfc with the current git version?
- Probably the problem is not mfoc, but, to be sure, could you check it reading the card with other device (pe. your Android)?
- Could you provide the files with problems (new-tag.mfc, old-tag.mfc, mifare-classic-keys.txt)?

I am not sure about the problem. This bug is with ACR, but the other bug is pn532, so the problem probably is not in the driver. Do you have other cards to write? Do you have other NFC devices? Could you check it in other computers?

Happy new year!
Regards,
kix

msschulte, 2021-01-02:
Hi!

I am using the RaspiZW with PN532 and ArchLinuxARM.
It is [currently](https://archlinuxarm.org/packages/arm/libnfc) the latest release libnfc-1.8.0 (f02ff51).

My test results:

----------------------------------------------------------------------

1. I wrote the following new image to a new/blank card:

```
00000000 77 01 73 2D │ 28 08 04 00 │ 62 63 64 65 │ 66 67 68 69
00000010 AA AA AA AA │ AA AA AA AA │ AA AA AA AA │ AA AA AA AA
00000020 AA AA AA AA │ AA AA AA AA │ AA AA AA AA │ AA AA AA AA
00000030 55 55 55 55 │ 55 55 0F 00 │ FF FF 55 55 │ 55 55 55 55
00000040 AA AA AA AA │ AA AA AA AA │ AA AA AA AA │ AA AA AA AA
00000050 AA AA AA AA │ AA AA AA AA │ AA AA AA AA │ AA AA AA AA
00000060 AA AA AA AA │ AA AA AA AA │ AA AA AA AA │ AA AA AA AA
00000070 55 55 55 55 │ 55 55 0F 00 │ FF FF 55 55 │ 55 55 55 55
...
000003C0 AA AA AA AA │ AA AA AA AA │ AA AA AA AA │ AA AA AA AA
000003D0 AA AA AA AA │ AA AA AA AA │ AA AA AA AA │ AA AA AA AA
000003E0 AA AA AA AA │ AA AA AA AA │ AA AA AA AA │ AA AA AA AA
000003F0 55 55 55 55 │ 55 55 0F 00 │ FF FF 55 55 │ 55 55 55 55
```

NFC reader: PN532 opened
Found MIFARE Classic card:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04  
       UID (NFCID1): 77  01  73  2d  
      SAK (SEL_RES): 08  
RATS support: no
Guessing size: seems to be a 1024-byte card
Writing 64 blocks |............................................................|
Done, 60 of 64 blocks written.

----------------------------------------------------------------------

2. I read the image for comparison (with new key):

NFC reader: PN532 opened
Found MIFARE Classic card:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04  
       UID (NFCID1): 77  01  73  2d  
      SAK (SEL_RES): 08  
RATS support: no
Guessing size: seems to be a 1024-byte card
Reading out 64 blocks |nfc_initiator_transceive_bytes: Mifare Authentication Failed!
Error: authentication failed for block 0x3f

FAILED, because the new keys/acs was not written.

----------------------------------------------------------------------

3. I read the image again for comparison (with old key):

NFC reader: PN532 opened
Found MIFARE Classic card:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04  
       UID (NFCID1): 77  01  73  2d  
      SAK (SEL_RES): 08  
RATS support: no
Guessing size: seems to be a 1024-byte card
Reading out 64 blocks |................................................................|
Done, 64 of 64 blocks read.
Writing data to file: test1.mfd ...Done.

Result:
```
00000000 77 01 73 2D │ 28 08 04 00 │ 62 63 64 65 │ 66 67 68 69
00000010 00 00 00 00 │ 00 00 00 00 │ 00 00 00 00 │ 00 00 00 00 <WRONG
00000020 00 00 00 00 │ 00 00 00 00 │ 00 00 00 00 │ 00 00 00 00 <WRONG
00000030 FF FF FF FF │ FF FF FF 07 │ 80 69 FF FF │ FF FF FF FF <WRONG
00000040 AA AA AA AA │ AA AA AA AA │ AA AA AA AA │ AA AA AA AA <CORRECT
00000050 00 00 00 00 │ 00 00 00 00 │ 00 00 00 00 │ 00 00 00 00 <WRONG
00000060 00 00 00 00 │ 00 00 00 00 │ 00 00 00 00 │ 00 00 00 00 <WRONG
00000070 FF FF FF FF │ FF FF FF 07 │ 80 69 FF FF │ FF FF FF FF <WRONG
...
000003C0 AA AA AA AA │ AA AA AA AA │ AA AA AA AA │ AA AA AA AA <CORRECT
000003D0 00 00 00 00 │ 00 00 00 00 │ 00 00 00 00 │ 00 00 00 00 <WRONG
000003E0 00 00 00 00 │ 00 00 00 00 │ 00 00 00 00 │ 00 00 00 00 <WRONG
000003F0 FF FF FF FF │ FF FF FF 07 │ 80 69 FF FF │ FF FF FF FF <WRONG
```

----------------------------------------------------------------------

4. I wrote the new image again with the current Git version 1.8.0 (7ebf9b9):

NFC reader: PN532 opened
Found MIFARE Classic card:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04  
       UID (NFCID1): 77  01  73  2d  
      SAK (SEL_RES): 08  
RATS support: no
Guessing size: seems to be a 1024-byte card
Writing 63 blocks |...............................................................|
Done, 63 of 64 blocks written.

----------------------------------------------------------------------

5. I read the image again for comparison (with new key):

NFC reader: PN532 opened
Found MIFARE Classic card:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04  
       UID (NFCID1): 77  01  73  2d  
      SAK (SEL_RES): 08  
RATS support: no
Guessing size: seems to be a 1024-byte card
Reading out 64 blocks |................................................................|
Done, 64 of 64 blocks read.
Writing data to file: test2.mfd ...Done.

Result:
```
00000000 77 01 73 2D │ 28 08 04 00 │ 62 63 64 65 │ 66 67 68 69
00000010 AA AA AA AA │ AA AA AA AA │ AA AA AA AA │ AA AA AA AA <CORRECT
00000020 AA AA AA AA │ AA AA AA AA │ AA AA AA AA │ AA AA AA AA <CORRECT
00000030 55 55 55 55 │ 55 55 0F 00 │ FF FF 55 55 │ 55 55 55 55 <CORRECT
00000040 AA AA AA AA │ AA AA AA AA │ AA AA AA AA │ AA AA AA AA <CORRECT
00000050 AA AA AA AA │ AA AA AA AA │ AA AA AA AA │ AA AA AA AA <CORRECT
00000060 AA AA AA AA │ AA AA AA AA │ AA AA AA AA │ AA AA AA AA <CORRECT
00000070 55 55 55 55 │ 55 55 0F 00 │ FF FF 55 55 │ 55 55 55 55 <CORRECT
...
000003C0 AA AA AA AA │ AA AA AA AA │ AA AA AA AA │ AA AA AA AA <CORRECT
000003D0 AA AA AA AA │ AA AA AA AA │ AA AA AA AA │ AA AA AA AA <CORRECT
000003E0 AA AA AA AA │ AA AA AA AA │ AA AA AA AA │ AA AA AA AA <CORRECT
000003F0 55 55 55 55 │ 55 55 0F 00 │ FF FF 55 55 │ 55 55 55 55 <CORRECT
```

----------------------------------------------------------------------

So, the latest Git version 1.8.0 (7ebf9b9) works for me.

I do not know what caused the problem,
but the release of 1.8.0 (f02ff51) on May 22, 2020
seems unusable for writing Mifare Classic cards.

thekix, 2021-01-02:
Thanks a lot @msschulte,

probably we can close this bug then. Is it ok for you?

Perhaps you should open an issue in the Arch Linux package to update it to the latest repo version.

Best regards,
kix

msschulte, 2021-01-02:
Yep :)

closed 2021-01-02 by smortex