nfc-tools/libnfc #631: pn532 (red board) no respond to card (i2c, SPI) is visible but nfc-poll does not scan

mumin50, 2020-11-28:
Hi im noob to NFC but trying to make it run. Is PN532 NFC RFID module V3 red PCB.
Managed to conect it to Raspberry Pi3B+ via all 3 posibilities... Module is visible, library talks to module (nfc-list), but when dooing nfc-poll it just not work. Tried 2 modules both same issue
I tried 
1)I2C it is there on 24, nfc-list looks ok  
2)SPI also lists 

example:
_pi@Qubikon:/etc/nfc $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- 24 -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
pi@Qubikon:/etc/nfc $

pi@Qubikon:/etc/nfc $ nfc-list -v
nfc-list uses libnfc 1.7.1
NFC device: pn532_i2c:/dev/i2c-1 opened
0 ISO14443A passive target(s) found.

0 Felica (212 kbps) passive target(s) found.

0 Felica (424 kbps) passive target(s) found.

0 ISO14443B passive target(s) found.

0 ISO14443B' passive target(s) found.

0 ISO14443B-2 ST SRx passive target(s) found.

0 ISO14443B-2 ASK CTx passive target(s) found.

0 Jewel passive target(s) found._

But nfc-poll gives timeout 
_pi@Qubikon:/etc/nfc $ nfc-poll
nfc-poll uses libnfc 1.7.1
NFC reader: pn532_i2c:/dev/i2c-1 opened
NFC device will poll during 30000 ms (20 pollings of 300 ms for 5 modulations)
nfc_initiator_poll_target: Success_

Any advise why...?

thekix, 2021-01-01:
Hi,

how are set the switches?

Regards,
kix

thekix, 2021-01-02:
FYI:

Check these pages:

http://wiki.sunfounder.cc/index.php?title=PN532_NFC_Module_for_Raspberry_Pi
https://components101.com/wireless/pn532-nfc-rfid-module

Cheers,
kix

FranckHillrom, 2022-12-06:
Hi, Someone have solution ?
I have the same output ...

pi@HomePi:~ $ sudo LIBNFC_LOG_LEVEL=3 nfc-poll
nfc-poll uses libnfc 1.8.0
debug   libnfc.config   key: [allow_autoscan], value: [true]
debug   libnfc.config   key: [allow_intrusive_scan], value: [false]
debug   libnfc.config   key: [log_level], value: [1]
debug   libnfc.config   key: [device.name], value: [PN532 over I2C]
debug   libnfc.config   key: [device.connstring], value: [pn532_i2c:/dev/i2c-1]
debug   libnfc.config   Unable to open directory: /etc/nfc/devices.d
debug   libnfc.general  log_level is set to 3
debug   libnfc.general  allow_autoscan is set to true
debug   libnfc.general  allow_intrusive_scan is set to false
debug   libnfc.general  1 device(s) defined by user
debug   libnfc.general    #0 name: "PN532 over I2C", connstring: "pn532_i2c:/dev/i2c-1"
debug   libnfc.chip.pn53x       Diagnose
debug   libnfc.chip.pn53x       Timeout value: 500
debug   libnfc.chip.pn53x       SAMConfiguration
debug   libnfc.chip.pn53x       Timeout value: 1000
debug   libnfc.bus.i2c  TX: 00 00 ff 03 fd d4 14 01 17 00
debug   libnfc.bus.i2c  wrote 10 bytes successfully.
debug   libnfc.chip.pn53x       PN53x ACKed
debug   libnfc.bus.i2c  TX: 00 00 ff 09 f7 d4 00 00 6c 69 62 6e 66 63 be 00
debug   libnfc.bus.i2c  wrote 16 bytes successfully.
debug   libnfc.chip.pn53x       PN53x ACKed
debug   libnfc.chip.pn53x       GetFirmwareVersion
debug   libnfc.bus.i2c  TX: 00 00 ff 02 fe d4 02 2a 00
debug   libnfc.bus.i2c  wrote 9 bytes successfully.
debug   libnfc.chip.pn53x       PN53x ACKed
debug   libnfc.chip.pn53x       SetParameters
debug   libnfc.bus.i2c  TX: 00 00 ff 03 fd d4 12 14 06 00
debug   libnfc.bus.i2c  wrote 10 bytes successfully.
debug   libnfc.chip.pn53x       PN53x ACKed
debug   libnfc.general  "PN532 over I2C" (pn532_i2c:/dev/i2c-1) has been claimed.
debug   libnfc.general  set_property_bool NP_ACTIVATE_FIELD False
debug   libnfc.chip.pn53x       ReadRegister
debug   libnfc.bus.i2c  TX: 00 00 ff 0c f4 d4 06 63 02 63 03 63 0d 63 38 63 3d b0 00
debug   libnfc.bus.i2c  wrote 19 bytes successfully.
debug   libnfc.chip.pn53x       PN53x ACKed
debug   libnfc.chip.pn53x       PN53X_REG_CIU_TxMode (Defines the transmission data rate and framing during transmission)
debug   libnfc.chip.pn53x       PN53X_REG_CIU_RxMode (Defines the transmission data rate and framing during receiving)
debug   libnfc.chip.pn53x       WriteRegister
debug   libnfc.bus.i2c  TX: 00 00 ff 08 f8 d4 08 63 02 80 63 03 80 59 00
debug   libnfc.bus.i2c  wrote 15 bytes successfully.
debug   libnfc.chip.pn53x       PN53x ACKed
debug   libnfc.chip.pn53x       RFConfiguration
debug   libnfc.bus.i2c  TX: 00 00 ff 04 fc d4 32 01 00 f9 00
debug   libnfc.bus.i2c  wrote 11 bytes successfully.
debug   libnfc.chip.pn53x       PN53x ACKed
debug   libnfc.general  set_property_bool NP_ACTIVATE_FIELD True
debug   libnfc.chip.pn53x       RFConfiguration
debug   libnfc.bus.i2c  TX: 00 00 ff 04 fc d4 32 01 01 f8 00
debug   libnfc.bus.i2c  wrote 11 bytes successfully.
debug   libnfc.chip.pn53x       PN53x ACKed
debug   libnfc.general  set_property_bool NP_INFINITE_SELECT True
debug   libnfc.chip.pn53x       RFConfiguration
debug   libnfc.bus.i2c  TX: 00 00 ff 06 fa d4 32 05 ff ff ff f8 00
debug   libnfc.bus.i2c  wrote 13 bytes successfully.
debug   libnfc.chip.pn53x       PN53x ACKed
debug   libnfc.general  set_property_bool NP_AUTO_ISO14443_4 True
debug   libnfc.general  set_property_bool NP_FORCE_ISO14443_A True
debug   libnfc.general  set_property_bool NP_FORCE_SPEED_106 True
debug   libnfc.general  set_property_bool NP_ACCEPT_INVALID_FRAMES False
debug   libnfc.general  set_property_bool NP_ACCEPT_MULTIPLE_FRAMES False
NFC reader: PN532 over I2C opened
NFC device will poll during 36000 ms (20 pollings of 300 ms for 6 modulations)
debug   libnfc.chip.pn53x       ReadRegister
debug   libnfc.bus.i2c  TX: 00 00 ff 0e f2 d4 06 63 02 63 03 63 05 63 38 63 3c 63 3d 19 00
debug   libnfc.bus.i2c  wrote 21 bytes successfully.
debug   libnfc.chip.pn53x       PN53x ACKed
debug   libnfc.chip.pn53x       PN53X_REG_CIU_TxAuto (Controls the settings of the antenna driver)
debug   libnfc.chip.pn53x       PN53X_REG_CIU_Control (Contains miscellaneous control bits)
debug   libnfc.chip.pn53x       WriteRegister
debug   libnfc.bus.i2c  TX: 00 00 ff 08 f8 d4 08 63 05 40 63 3c 10 cd 00
debug   libnfc.bus.i2c  wrote 15 bytes successfully.
debug   libnfc.chip.pn53x       PN53x ACKed
debug   libnfc.chip.pn53x       InAutoPoll
debug   libnfc.chip.pn53x       No timeout
debug   libnfc.bus.i2c  TX: 00 00 ff 0b f5 d4 60 14 02 20 10 03 11 12 04 03 59 00
debug   libnfc.bus.i2c  wrote 18 bytes successfully.
debug   libnfc.chip.pn53x       PN53x ACKed
No target found.
debug   libnfc.chip.pn53x       InRelease
debug   libnfc.bus.i2c  TX: 00 00 ff 03 fd d4 52 00 da 00
debug   libnfc.bus.i2c  wrote 10 bytes successfully.
debug   libnfc.chip.pn53x       PN53x ACKed
debug   libnfc.general  set_property_bool NP_ACTIVATE_FIELD False
debug   libnfc.chip.pn53x       RFConfiguration
debug   libnfc.bus.i2c  TX: 00 00 ff 04 fc d4 32 01 00 f9 00
debug   libnfc.bus.i2c  wrote 11 bytes successfully.
debug   libnfc.chip.pn53x       PN53x ACKed
debug   libnfc.chip.pn53x       PowerDown
debug   libnfc.bus.i2c  TX: 00 00 ff 03 fd d4 16 f0 26 00
debug   libnfc.bus.i2c  wrote 10 bytes successfully.
debug   libnfc.chip.pn53x       PN53x ACKed