nfc-tools/libnfc #632: nfc-mfclassic write bugged

TrixyCracker, 2020-12-11:
Often when i try to write a tag, shows "Authentication failed", while i'm able to write the card, the program writes 60/64 block(wtf?).
Obv, all is working perfectly on 1.7.1. Pls fix this, im not the only one have this problem.
[image]

thekix, 2021-01-02:
Hello @TrixyCracker 

could you provide the command used?

Regards,
kix

msschulte, 2021-01-03:
@TrixyCracker, please take a look at #629. If possible use the newest Git version.

@thekix, '60/64 blocks written' is an indication of version 1.8.0 f02ff51 .

nezzzumi, 2023-02-15:
Building from scratch works for me.

Thanks!!