nfc-tools/libnfc #634: PCSC "Insufficient Buffer" error return while using ifdnfc-activate

ronie19, 2021-01-21:
https://github.com/nfc-tools/libnfc/blob/7ebf9b92d687a53ec156e809517beb97bd5bf59d/libnfc/drivers/pcsc.c#L604

Using Raspberry Pi
Gives this error. Libnfc is installed and nfc-poll and nfc-list commands work properly.
ifdnfc-activate gives off this error that I have traced back to pcsc and SCardGetAttrib() .
Any leads to increase buffer size ?

eximius313, 2022-01-29:
@ronie19 I have exactly the same problem :(

`nfc-scan-device` gives me:
```
nfc-scan-device uses libnfc libnfc-1.8.0-63-g3df7f25
1 NFC device(s) found:
- _PN532_UART:
    pn532_uart:/dev/ttyUSB0:115200
```

`nfc-poll` gives:
```
nfc-poll uses libnfc libnfc-1.8.0-63-g3df7f25
NFC reader: _PN532_UART opened
NFC device will poll during 36000 ms (20 pollings of 300 ms for 6 modulations)
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04
       UID (NFCID1): f2  4b  14  1b
      SAK (SEL_RES): 08
```

But `ifdnfc-activate` gives:
```
Activating ifdnfc with "pn532_uart:/dev/ttyUSB0:115200"...
Insufficient buffer.
```

and `pcsc_scan` **does not see any card**:
```
Using reader plug'n play mechanism
Scanning present readers...
0: _PN532_UART 00 00

 Reader 0: _PN532_UART 00 00
  Event number: 0
  Card state: Card removed,
```

My config is:
`/etc/nfc/libnfc.conf`
```
allow_autoscan = true
allow_intrusive_scan = false
log_level = 1
device.name = "_PN532_UART"
device.connstring = "pn532_uart:/dev/ttyUSB0:115200"
```

and `/etc/reader.conf.d/libifdnfc` is:
```
FRIENDLYNAME      "_PN532_UART"
DEVICENAME        /dev/ttyUSB0
LIBPATH           /usr/local/lib/libifdnfc.so
CHANNELID         1
```

Have you managed to solve this issue?

Kind regards

ronie19, 2022-01-29:
> @ronie19 I have exactly the same problem :(
>
> `nfc-scan-device` gives me:
>
> ```
> nfc-scan-device uses libnfc libnfc-1.8.0-63-g3df7f25
> 1 NFC device(s) found:
> - _PN532_UART: [...]

Yes... Only solution I have found is using 64-bit raspberry Pi OS ...32-bit OS gives buffer problem.

eximius313, 2022-02-02:
Thanks, on aarch64 it works indeed.
Too bad that 32bit is still affected :(

b4wllyb4g5, 2024-04-02:
Hello, sorry to resurrect an old thread.  I am having exactly same problem as you have described.

I am getting insufficient buffer when trying to run `ifdnfc-activate`.

I am using a Raspberry Pi Zero 2 W and a PN532 (Red model) connected on the i2c bus.

`nfc-poll` is working fine, and also `pcsc_scan` works, just i am unable to run `ifdnfc-activate` with the error

`Activating ifdnfc with "pn532_i2c:/dev/i2c-1"...
Insufficient buffer.`

What model of Raspberry Pi and distro did you use to end up getting the reader to work fully? Thank you

ronie19, 2024-04-02:
> Hello, sorry to resurrect an old thread. I am having exactly same problem as you have described.
>
> I am getting insufficient buffer when trying to run `ifdnfc-activate`.
>
> I am using a Raspberry Pi [...]

Simply use 64 bit RPi OS. The latest legacy version only.

b4wllyb4g5, 2024-04-02:
thanks for the quick reply.

I tried using "kali-linux-2024.1-raspberry-pi-arm64-img-xz" but i was unable to compile ifdnfc i kept getting an error when running 

`autoreconf -vis
./configure`

I believe it was about not finding nfc.h file and was aborting.  I was using the same install method as when on a 32bit distro.