nfc-tools/libnfc #646: error libnfc.driver.acr122_usb Wrong reply

amrbt, 2021-06-22:
Hello!

I'm stucked at this point with the libnfc 1.8.0 and do not find information about this error:

> **nfc-list -v                                                        130 ⨯**
> nfc-list uses libnfc 1.8.0
> NFC device: ACS / ACR122U PICC Interface opened
> 1 ISO14443A passive target(s) found:
> ISO/IEC 14443A (106 kbps) target:
>     ATQA (SENS_RES): 00  04  
> * UID size: single
> * bit frame anticollision supported
>        UID (NFCID1): 67  32  64  c3  
>       SAK (SEL_RES): 08  
> * Not compliant with ISO/IEC 14443-4
> * Not compliant with ISO/IEC 18092
> 
> Fingerprinting based on MIFARE type Identification Procedure:
> * MIFARE Classic 1K
> * MIFARE Plus (4 Byte UID or 4 Byte RID) 2K, Security level 1
> * SmartMX with MIFARE 1K emulation
> Other possible matches based on ATQA & SAK values:
> 
> error   libnfc.driver.acr122_usb        Wrong reply
> 0 Felica (212 kbps) passive target(s) found.
> 
> error   libnfc.driver.acr122_usb        Wrong reply
> 0 Felica (424 kbps) passive target(s) found.
> 
> error   libnfc.driver.acr122_usb        Wrong reply
> 0 ISO14443B passive target(s) found.
> 
> 0 ISO14443B' passive target(s) found.
> 
> 0 ISO14443B-2 ST SRx passive target(s) found.
> 
> 0 ISO14443B-2 ASK CTx passive target(s) found.
> 
> 0 ISO14443B iClass passive target(s) found.
> 
> error   libnfc.driver.acr122_usb        Wrong reply
> 0 ISO14443A-3 Jewel passive target(s) found.
> 
> 0 ISO14443A-2 NFC Barcode passive target(s) found.
> 
> error   libnfc.driver.acr122_usb        Wrong reply
> 

> **cat /etc/os-release** 
> PRETTY_NAME="Kali GNU/Linux Rolling"
> NAME="Kali GNU/Linux"
> ID=kali
> VERSION="2021.2"
> VERSION_ID="2021.2"
> VERSION_CODENAME="kali-rolling"
> ID_LIKE=debian
> ANSI_COLOR="1;31"
> HOME_URL="https://www.kali.org/"
> SUPPORT_URL="https://forums.kali.org/"
> BUG_REPORT_URL="https://bugs.kali.org/"
> 
> **uname -r**
> 5.10.0-kali7-amd64

> **lsusb -v**    
> Bus 002 Device 005: ID 072f:2200 Advanced Card Systems, Ltd ACR122U
> Device Descriptor:
>   bLength                18
>   bDescriptorType         1
>   bcdUSB               1.10
>   bDeviceClass            0 
>   bDeviceSubClass         0 
>   bDeviceProtocol         0 
>   bMaxPacketSize0        64
>   idVendor           0x072f Advanced Card Systems, Ltd
>   idProduct          0x2200 ACR122U
>   bcdDevice            2.14
>   iManufacturer           1 ACS
>   iProduct                2 ACR122U PICC Interface
>   iSerial                 0 
>   bNumConfigurations      1
>   Configuration Descriptor:
>     bLength                 9
>     bDescriptorType         2
>     wTotalLength       0x005d
>     bNumInterfaces          1
>     bConfigurationValue     1
>     iConfiguration          0 
>     bmAttributes         0x80
>       (Bus Powered)
>     MaxPower              200mA
>     Interface Descriptor:
>       bLength                 9
>       bDescriptorType         4
>       bInterfaceNumber        0
>       bAlternateSetting       0
>       bNumEndpoints           3
>       bInterfaceClass        11 Chip/SmartCard
>       bInterfaceSubClass      0 
>       bInterfaceProtocol      0 
>       iInterface              0 
>       ChipCard Interface Descriptor:
>         bLength                54
>         bDescriptorType        33
>         bcdCCID              1.00
>         nMaxSlotIndex           0
>         bVoltageSupport         7  5.0V 3.0V 1.8V 
>         dwProtocols             2  T=1
>         dwDefaultClock       4000
>         dwMaxiumumClock      4000
>         bNumClockSupported      0
>         dwDataRate          10752 bps
>         dwMaxDataRate      250000 bps
>         bNumDataRatesSupp.      0
>         dwMaxIFSD             256
>         dwSyncProtocols  00000000 
>         dwMechanical     00000000 
>         dwFeatures       00020040
>           Auto parameter negotiation made by CCID
>           Short APDU level exchange
>         dwMaxCCIDMsgLen       271
>         bClassGetResponse      00
>         bClassEnvelope         00
>         wlcdLayout           none
>         bPINSupport             0 
>         bMaxCCIDBusySlots       1
>       Endpoint Descriptor:
>         bLength                 7
>         bDescriptorType         5
>         bEndpointAddress     0x81  EP 1 IN
>         bmAttributes            3
>           Transfer Type            Interrupt
>           Synch Type               None
>           Usage Type               Data
>         wMaxPacketSize     0x0008  1x 8 bytes
>         bInterval              50
>       Endpoint Descriptor:
>         bLength                 7
>         bDescriptorType         5
>         bEndpointAddress     0x04  EP 4 OUT
>         bmAttributes            2
>           Transfer Type            Bulk
>           Synch Type               None
>           Usage Type               Data
>         wMaxPacketSize     0x0040  1x 64 bytes
>         bInterval               0
>       Endpoint Descriptor:
>         bLength                 7
>         bDescriptorType         5
>         bEndpointAddress     0x83  EP 3 IN
>         bmAttributes            2
>           Transfer Type            Bulk
>           Synch Type               None
>           Usage Type               Data
>         wMaxPacketSize     0x0040  1x 64 bytes
>         bInterval               0
> can't get debug descriptor: Resource temporarily unavailable
> Device Status:     0x0000
>   (Bus Powered)

davidebeatrici, 2023-04-15:
Possibly related to #666.

<details>
<summary>lsusb</summary>

```
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x072f Advanced Card Systems, Ltd
  idProduct          0x2200 ACR122U
  bcdDevice            2.14
  iManufacturer           1 ACS
  iProduct                2 ACR122U PICC Interface
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x005d
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              200mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass        11 Chip/SmartCard
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      ChipCard Interface Descriptor:
        bLength                54
        bDescriptorType        33
        bcdCCID              1.00
        nMaxSlotIndex           0
        bVoltageSupport         7  5.0V 3.0V 1.8V 
        dwProtocols             2  T=1
        dwDefaultClock       4000
        dwMaxiumumClock      4000
        bNumClockSupported      0
        dwDataRate          10752 bps
        dwMaxDataRate      250000 bps
        bNumDataRatesSupp.      0
        dwMaxIFSD             256
        dwSyncProtocols  00000000 
        dwMechanical     00000000 
        dwFeatures       00020040
          Auto parameter negotiation made by CCID
          Short APDU level exchange
        dwMaxCCIDMsgLen       271
        bClassGetResponse      00
        bClassEnvelope         00
        wlcdLayout           none
        bPINSupport             0 
        bMaxCCIDBusySlots       1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              50
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
```
</details>

davidebeatrici, 2023-04-16:
Please note that the device appears to be working just fine with `pcscd` and https://github.com/Flowtter/py-acr122u.

davidebeatrici, 2023-04-21:
I acquired another seemingly identical device, this time with a mini CD included.

Here's the SDK that is provided: [ACR122U_SDK.zip](https://github.com/nfc-tools/libnfc/files/11298661/ACR122U_SDK.zip)

dkdk22, 2023-12-11:
Is the mentioned error resolved? I'm experiencing the same issue.

Gapoly, 2024-02-26:
Same problem here. I've tried using my ACR122U on 4 diffirents OS, 2 different machines:

- Kali 2023.4
- Debian 12
- Ubuntu Server 22.04.4 LTS
- Kubuntu 22.04.4 LTS

All with the same problem, "error libnfc.driver.acr122_usb Wrong reply"
Hopefully the problem comes from the device and getting another one will fix my issue.

Gapoly, 2024-02-29:
Update: I've bought a new one and I don't have the error anymore. Either the other guy sold me a fake or it was damaged.

davidebeatrici, 2024-02-29:
Could you send here a picture of the device's back, please?

Gapoly, 2024-03-03:
> Could you send here a picture of the device's back, please?

Same back as the old one I had
[image]

davidebeatrici, 2024-03-03:
Interesting... in that case there may be no easy way to determine whether a specific reader is going to work.

italodeveloper, 2024-11-13:
> > Could you send here a picture of the device's back, please?
> 
> Same back as the old one I had ![IMG_20240303_153957](https://private-user-images.githubusercontent.com/94485066/309557755-57850237-8862-4a72-a415-f9d7301e96f4.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE0NjA3MDYsIm5iZiI6MTczMTQ2MDQwNiwicGF0aCI6Ii85NDQ4NTA2Ni8zMDk1NTc3NTUtNTc4NTAyMzctODg2Mi00YTcyLWE0MTUtZjlkNzMwMWU5NmY0LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTEzVDAxMTMyNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTEwOWQ1ODE3MDFjOWJmOGZjZmU3NDgzMTA5OTFjY2E3NTVkMWY0NTVkYWM1ZjJjMzZmOTFmZTRhN2Y2ODBhNDQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.iIYLcGg3NFECuOcvX5LnFUDQOrcv_GaFuT_ldd1xJiU)

Is this the device working or defective? I bought a defective one the serial number starts with RR171-XX

davidebeatrici, 2024-11-13:
My non-working device's serial number begins with `RR171-` too (https://github.com/nfc-tools/libnfc/issues/646#issuecomment-1509407184).

Now that I'm rereading the comments I realized @Gapoly's one is unclear. Back when I replied I assumed he meant the fake/defective had the same SN prefix.

italodeveloper, 2024-11-13:
> O número de série do meu dispositivo que não funciona começa com `RR171-`( [#646 (comentário)](https://github.com/nfc-tools/libnfc/issues/646#issuecomment-1509407184) ).
> 
> Agora que estou relendo os comentários percebi@Gapoly's one não está claro. Quando respondi, presumi que ele queria dizer que o falso/defeituoso tinha o mesmo prefixo SN.

That's right, my reader with the problem is also the **RR171-XX**. This reader that he posted in the photo is not clear if it has the problem or if it was the one that worked, I got it with the same initial serial number to buy **RR545-XX** but it didn't I wanted to waste time, if I answer if this RR545 model works I would be grateful

davidebeatrici, 2024-11-13:
Please note that I have a working device also beginning with `RR171-`.

Gapoly, 2024-11-20:
I just recently went back to NFC and I found the problem. Yes, I had a bad ACR then bought another one which works. The main problem is that I bought some Mifare Classic 1K Gen1 (bad product from aliexpress), you can't change UID on Gen1.

I bought some Gen2 that works on Mifare Classic Tool only and it works, I can change the UID. I'm looking for some Mifare Classic 1K Gen2 that works with nfc-mfclassic. Not a lot of choice in Europe.