nfc-tools/libnfc #656: What NFC readers are supported?

apsigma, 2021-09-22:
Is `ACR122u` the only supported NFC reader? The docs don't mention anything about supported devices, would be good.

I have a `ACR1255u-j1` that works on my Windows machine, but I have not been able to get it to work on Ubuntu 20.4 with libnfc, it says it cannot find the device. Is this expected?

```
>> nfc-list
nfc-list uses libnfc 1.8.0
No NFC device found.
```

cheriimoya, 2021-10-05:
i doubt this reader will work. If i understood it correctly, libnfc needs the reader hardware to support direct transmit. This is not the case for my ACR1252, so i have to use PC/SC.

mdxs, 2021-10-31:
Recently experimented with a DL533N CS - LibNFC NFC Reader from Digital Logic and using `nfcpy` (Python module using LibNFC) my code worked without modifications; in my configuration (using a Raspbeery Pi) I had to allow access to the new USB device and changed the hint to find it by replacing the Vendor ID and Product ID parts in the "usb:vid:pid" hint. However, initially I just changed "usb:vid:pid" to "usb" so that I could even swap and test the (old) AC122U and the (new) DL533N reader with my software. Found no issues so far, but YMMV.

verliba, 2023-10-17:
Hi, has anyone found a solution for writing to ACR1255U-J1 ?
In my case it does not even get found, just like for @apsigma
My aim ti so use it to write cards. Is it possible with libnfc, or what tools should I use on linux?
Thanks

neomilium, 2023-10-21:
Have a look to README file. AFAICR, supported devices are listed.

closed 2023-10-21 by neomilium