nfc-tools/libnfc #657: nfc-emulate-forum-tag4: ERROR: Operation not supported (95)

abrilevskiy, 2021-10-01:
>sudo nfc-emulate-forum-tag4 

NFC device: pn532_spi:/dev/spidev3.0 opened
Emulating NDEF tag now, please touch it with a second NFC device
    In: 00  a4  04  00  07  d2  76  00  00  85  01  01  00  
    Out: 90  00  
    In: 00  a4  00  0c  02  e1  03  
    Out: 90  00  
    In: 00  b0  00  00  0f  
    Out: 00  0f  20  00  54  00  ff  04  06  e1  04  ff  fe  00  00  90  00  
    In: 00  a4  00  0c  02  e1  04  
    Out: 90  00  
    In: 00  b0  00  00  02  
    Out: 00  21  90  00  
    In: 00  b0  00  02  21  
    Out: d1  02  1c  53  70  91  01  09  54  02  65  6e  4c  69  62  6e  66  63  51  01  0b  55  03  6c  69  62  6e  66  63  2e  6f  72  67  90  00  
    In: 60  
nfc-emulate-forum-tag4: ERROR: Operation not supported (95)
nfc_emulate_target: Success

AdamLaurie, 2021-10-02:
please confirm tag type you're trying to emulate - looks like the reader is trying to initiate a Mifare authentication which wouldn't be appropriate for this type of emulation.

iceman1001, 2021-10-02:
doesn't that 60 need to be accompanied with a blockno ?  its only one byte here.   A bit of an odd output.

abrilevskiy, 2021-10-03:
I need to emulate Mifare Classic tag - I did not found anything ready to use yet and explore how nfc-emulate-forum-tag4 works. The nfc-emulate-forum-tag4 emulate Mifare Plus - as a reader I use NFC Tool Android application.
I tried 1.7 release (from repository) and 1.8 builded myself - the issue is reproduced.
Please let me know what information can help to identify the issue - I will do my best to provide.

iceman1001, 2021-10-03:
Then emulating a NFC/Type 4 isn't gonna work.  Rather sounds like it is working as expected.