nfc-tools/libnfc #658: ACR1252U is still not recognized.

paranoid64, 2021-10-06:
ACR1252U is still not recognized, but should be supported in Version 1.8.0

DMESG:

Device= 1.10
[   68.994259] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   68.994261] usb 3-1: Product: ACR1252 Dual Reader
[   68.994263] usb 3-1: Manufacturer: ACS
[  114.371920] usb 3-2: new full-speed USB device number 7 using xhci_hcd
[  114.635324] usb 3-2: New USB device found, idVendor=072f, idProduct=2200, bcdDevice= 2.14

[  114.635327] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  114.635328] usb 3-2: Product: ACR122U PICC Interface
[  114.635330] usb 3-2: Manufacturer: ACS
[  127.243529] usb 3-2: reset full-speed USB device number 7 using xhci_hcd
[  135.951669] usb 3-2: reset full-speed USB device number 7 using xhci_hcd

$ nfc-list
nfc-list uses libnfc libnfc-1.8.0-41-gd9a04a5
NFC device: ACS / ACR122U PICC Interface opened

Don't be surprised, I have 1 x ACR1252 and 1 x ACR122U plugged in. But only the ACR122U is recognized.

paranoid64, 2021-10-07:
LIBNFC_LOG_LEVEL=3 nfc-list -v
info	libnfc.config	Unable to open file: /usr/local/etc/nfc/libnfc.conf
debug	libnfc.config	Unable to open directory: /usr/local/etc/nfc/devices.d
debug	libnfc.general	log_level is set to 3
debug	libnfc.general	allow_autoscan is set to true
debug	libnfc.general	allow_intrusive_scan is set to false
debug	libnfc.general	0 device(s) defined by user
nfc-list uses libnfc libnfc-1.8.0-41-gd9a04a5
debug	libnfc.general	0 device(s) found using acr122_usb driver
debug	libnfc.general	0 device(s) found using pn53x_usb driver
No NFC device found.

the configs are expected somewhere else entirely. According to the instructions, /etc/nfc is specified, for example.
The Debug info show  /usr/local/etc/nfc/ ?

VictoCen, 2023-01-09:
acsccid-1.1.8 is compiled and installed

victor@deb11:~$ uname -r
5.10.0-20-amd64

victor@deb11:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 004: ID 072f:223b Advanced Card Systems, Ltd ACR1252 Dual Reader
Bus 001 Device 003: ID 2109:2812 VIA Labs, Inc. VL812 Hub
Bus 001 Device 002: ID 2109:2812 VIA Labs, Inc. VL812 Hub
Bus 001 Device 008: ID 072f:2200 Advanced Card Systems, Ltd ACR122U
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

victor@deb11:~$nfc-list
nfc-list uses libnfc 1.8.0
NFC device: ACS / ACR122U PICC Interface opened

victor@deb11:~$nfc-scan-device 
nfc-scan-device uses libnfc 1.8.0
1 NFC device(s) found:
- ACS / ACR122U PICC Interface:
    acr122_usb:001:008

victor@deb11:~$ pcsc_scan
Using reader plug'n play mechanism
Scanning present readers...
0: ACS ACR1252 1S CL Reader [ACR1252 Dual Reader PICC] 00 00
1: ACS ACR1252 1S CL Reader [ACR1252 Dual Reader SAM] 01 00
2: ACS ACR122U 02 00
 
Mon Jan  9 09:27:25 2023
 Reader 0: ACS ACR1252 1S CL Reader [ACR1252 Dual Reader PICC] 00 00
  Event number: 0
  Card state: Card removed, 
 Reader 1: ACS ACR1252 1S CL Reader [ACR1252 Dual Reader SAM] 01 00
  Event number: 1
  Card state: Card removed, 
 Reader 2: ACS ACR122U 02 00
  Event number: 0
  Card state: Card removed, 
 \

mariolpantunes, 2023-01-09:
@VictoCen sorry I did not fully understand, did libnfc worked with ACR152u?

VictoCen, 2023-01-09:
@mariolpantunes I'm not so familar with lib stuff but after deinstall of
libnfc6, libnfc-bin, libnfc-dev, libnfc-examples the commands 
pcsc_scan and opensc-tool --list-readers still detects the devices,
so I would point to xavave's message: Does not work with acr1252

HamadiSan, 2023-01-12:
[image]

HamadiSan, 2023-01-12:
So, nfc-list now detects ACR1252u after "sudo apt-get install libacr38u" on raspberry-pi.

edent, 2024-02-19:
I've tried this with a 1252u on Linux and installing [`libacr38u`](https://old-releases.ubuntu.com/ubuntu/pool/universe/a/acr38/?C=M;O=D).  It just shows "No NFC device found."

I can see the device with `pcsc_scan`

Has anyone got this working?

ottobaer, 2024-04-26:
It works through pcscd, but I never got any nfc tools working with it.
I think replacing the reader with a supported one is the best thing to do here.