nfc-tools/libnfc #666: error libnfc.driver.acr122_usb Command Code verification failed

Pebrok, 2022-01-15:
Hi,
I'm new to Libnfc. 
I'd like to use my ACR122U to read (or more) badges and cards, and I installed Kali on a USB stick. 
I start it on a pc on boot, and I installed everything, and it seemed to be working.

But everytime I do a nfc list , I got the following error message: 
error libnfc.driver.acr122_usb Command Code verification failed

I believe I looked everywhere on the net, and tried a lot of thing, but couldn't find a fix. 

Can someone help ? 
As I am a newbie, I copy/paste code, so if anything more complicated is requested, please be patient.

Gnu-Hack, 2022-11-06:
Hi!!

Gnu-Hack, 2022-11-06:
What command do you use?

phodina, 2022-11-07:
Hi @Gnu-Hack ,
I've encountered similar behaviour.

The NFC reader/writer is ACR122U-A9 based on the P/N.

I'm porting this package to GNU Guix. I tried both GNU and CMake systems. The package builds in both but as I attempt to run the following command I get this error:
```
# herd status pcscd
...
Stopped:
 - pcscd
...
$ lsusb
...
Bus 001 Device 101: ID 072f:2200 Advanced Card Systems, Ltd ACR122U
...
# /gnu/store/f8pq1slc44lldfnnbhfnxq66aivwmav2-libnfc-1.8.0/bin/nfc-list
/gnu/store/f8pq1slc44lldfnnbhfnxq66aivwmav2-libnfc-1.8.0/bin/nfc-list uses libnfc 1.8.0
NFC device: ACS / ACR122U PICC Interface opened
error	libnfc.driver.acr122_usb	Command Code verification failed
error	libnfc.driver.acr122_usb	Command Code verification failed
error	libnfc.driver.acr122_usb	Command Code verification failed
error	libnfc.driver.acr122_usb	Command Code verification failed
error	libnfc.driver.acr122_usb	Command Code verification failed
error	libnfc.driver.acr122_usb	Command Code verification failed
error	libnfc.driver.acr122_usb	Command Code verification failed
error	libnfc.driver.acr122_usb	Command Code verification failed
error	libnfc.driver.acr122_usb	Command Code verification failed
error	libnfc.driver.acr122_usb	Command Code verification failed
error	libnfc.driver.acr122_usb	Command Code verification failed
error	libnfc.driver.acr122_usb	Command Code verification failed
$ lsmod
Module                  Size  Used by
pn533                  45056  0
nfc                   122880  1 pn533
...
```
I've tried to blacklist the modules but that didn't help.

Also is it normal I hear some sound when the device activates? My guess is it's charging the coil or something like that.
After that the tiny noise disappears and the LED on the device is turned on (red/amber).

Gnu-Hack, 2022-11-07:
Hi!!
have you installed the pccd drivers?
what have you blacklisted?
If the device has a red LED, it means that it is not correctly configured

José CardeLL

-------- Mensaje original --------

phodina, 2022-11-07:
Here's also the `dmesg`
```
[143292.130231] usb 1-3.3: new full-speed USB device number 103 using xhci_hcd
[143292.280946] usb 1-3.3: New USB device found, idVendor=072f, idProduct=2200, bcdDevice= 2.14
[143292.280960] usb 1-3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[143292.280966] usb 1-3.3: Product: ACR122U PICC Interface
[143292.280971] usb 1-3.3: Manufacturer: ACS
[143295.679311] usb 1-3.3: NFC: NXP PN532 firmware ver 1.6 now attached
[143344.473036] pn533_usb 1-3.3:1.0: NFC: NXP PN533 NFC device disconnected
[143344.602653] usb 1-3.3: reset full-speed USB device number 103 using xhci_hcd
[143344.857382] usb 1-3.3: NFC: NXP PN532 firmware ver 1.6 now attached
[143366.956943] usbcore: deregistering interface driver pn533_usb
[143366.957007] pn533_usb 1-3.3:1.0: NFC: NXP PN533 NFC device disconnected
[143370.078518] usb 1-3.3: reset full-speed USB device number 103 using xhci_hcd
```

I'll check the pccd drivers

Gnu-Hack, 2022-11-07:
Install you (Debian)

sudo apt-get install pcscd pcsc-tools

After:

sudo nano /etc/modprobe.d/blacklist-libnfc.conf

And write here:

blacklist nfc
blacklist pn533
blacklist pn533_usb

After do not restart, turn off and on

if the red led has gone off it is good.
then check with the following:

sudo nfc-list -v
nfc-scan-device
nfc-list
pcsc_scan

José CardeLL

-------- Mensaje original --------

phodina, 2022-11-07:
Okay, I'll test it on Debain tomorrow to see if the issue is the device itself or me packaging for GNU Guix.

However, these steps I already applied. But it might be the way how Guix is structured that it might need some patch.

Nevertheless thanks for the support!

Gnu-Hack, 2022-11-07:
OK we talk tomorrow

José CardeLL

-------- Mensaje original --------

Gnu-Hack, 2022-11-08:
hello, have you managed to do it?

José CardeLL

-------- Mensaje original --------

phodina, 2022-11-09:
Hi, yes I ran the commands on Fedora.

The `libnfc` does not work there either.

However I've been able to read the cards using `pcsc_scan` after blacklisting the modules.

Therefore the NFC reader/writer device is functional.

Regarding the LED. It stays Red the whole time. When reading card it turns Green and after it goes back to Red.

Here are some logs from GNU Guix. 

```
# /gnu/store/84vh7mcfgdpp19m1bwnancfr1ni369gs-libnfc-1.8.0/bin/nfc-list
/gnu/store/84vh7mcfgdpp19m1bwnancfr1ni369gs-libnfc-1.8.0/bin/nfc-list uses libnfc 1.8.0
error	libnfc.driver.acr122_usb	Unable to claim USB interface (Device or resource busy)
nfc-list: ERROR: Unable to open NFC device: acr122_usb:001:126
```

```
$ lsmod
Module                  Size  Used by
pn533_usb              20480  0
pn533                  45056  1 pn533_usb
nfc                   122880  1 pn533
```

```
rmmod pn533_usb pn533 nfc
```

```
# /gnu/store/84vh7mcfgdpp19m1bwnancfr1ni369gs-libnfc-1.8.0/bin/nfc-list
/gnu/store/84vh7mcfgdpp19m1bwnancfr1ni369gs-libnfc-1.8.0/bin/nfc-list uses libnfc 1.8.0
NFC device: ACS / ACR122U PICC Interface opened
error	libnfc.driver.acr122_usb	Command Code verification failed
error	libnfc.driver.acr122_usb	Command Code verification failed
error	libnfc.driver.acr122_usb	Command Code verification failed
error	libnfc.driver.acr122_usb	Command Code verification failed
error	libnfc.driver.acr122_usb	Command Code verification failed
error	libnfc.driver.acr122_usb	Command Code verification failed
error	libnfc.driver.acr122_usb	Command Code verification failed
error	libnfc.driver.acr122_usb	Command Code verification failed
error	libnfc.driver.acr122_usb	Command Code verification failed
error	libnfc.driver.acr122_usb	Command Code verification failed
error	libnfc.driver.acr122_usb	Command Code verification failed
error	libnfc.driver.acr122_usb	Command Code verification failed
```

```
# LIBNFC_LOG_LEVEL=3 /gnu/store/84vh7mcfgdpp19m1bwnancfr1ni369gs-libnfc-1.8.0/bin/nfc-list
info	libnfc.config	Unable to open file: /gnu/store/84vh7mcfgdpp19m1bwnancfr1ni369gs-libnfc-1.8.0/etc/nfc/libnfc.conf
debug	libnfc.config	Unable to open directory: /gnu/store/84vh7mcfgdpp19m1bwnancfr1ni369gs-libnfc-1.8.0/etc/nfc/devices.d
debug	libnfc.general	log_level is set to 3
debug	libnfc.general	allow_autoscan is set to true
debug	libnfc.general	allow_intrusive_scan is set to false
debug	libnfc.general	0 device(s) defined by user
/gnu/store/84vh7mcfgdpp19m1bwnancfr1ni369gs-libnfc-1.8.0/bin/nfc-list uses libnfc 1.8.0
debug	libnfc.driver.acr122_usb	device found: Bus 001 Device 126 Name ACS ACR122
debug	libnfc.general	1 device(s) found using acr122_usb driver
debug	libnfc.general	0 device(s) found using pn53x_usb driver
debug	libnfc.driver.acr122_usb	3 element(s) have been decoded from "acr122_usb:001:126"
debug	libnfc.driver.acr122_usb	TX: 62 00 00 00 00 00 00 01 00 00
debug	libnfc.driver.acr122_usb	RX: 80 02 00 00 00 00 00 00 81 00 3b 00
debug	libnfc.driver.acr122_usb	ACR122 PICC Operating Parameters
debug	libnfc.driver.acr122_usb	TX: 6f 05 00 00 00 00 00 00 00 00 ff 00 51 00 00
debug	libnfc.driver.acr122_usb	RX: 80 02 00 00 00 00 00 00 81 00 90 00
debug	libnfc.chip.pn53x	GetFirmwareVersion
debug	libnfc.driver.acr122_usb	TX: 6f 07 00 00 00 00 00 00 00 00 ff 00 00 00 02 d4 02
debug	libnfc.driver.acr122_usb	RX: 80 08 00 00 00 00 00 00 81 00 d5 03 32 01 06 07 90 00
[... 218 lines cut ...]
error	libnfc.driver.acr122_usb	Command Code verification failed
debug	libnfc.general	set_property_bool NP_HANDLE_CRC True
debug	libnfc.general	set_property_bool NP_HANDLE_PARITY True
debug	libnfc.driver.acr122_usb	ACR122 Abort
debug	libnfc.driver.acr122_usb	TX: 6f 07 00 00 00 00 00 00 00 00 ff 00 00 00 02 d4 02
debug	libnfc.driver.acr122_usb	RX: 80 04 00 00 00 00 00 00 81 00 d5 33 90 00
debug	libnfc.chip.pn53x	ReadRegister
debug	libnfc.driver.acr122_usb	TX: 6f 0d 00 00 00 00 00 00 00 00 ff 00 00 00 08 d4 06 63 02 63 03 63 0d
debug	libnfc.driver.acr122_usb	RX: 80 07 00 00 00 00 00 00 81 00 d5 07 83 0b 10 90 00
debug	libnfc.chip.pn53x	PN53X_REG_CIU_RxMode (Defines the transmission data rate and framing during receiving)
debug	libnfc.chip.pn53x	PN53X_REG_CIU_ManualRCV (Allows manual fine tuning of the internal receiver)
debug	libnfc.chip.pn53x	WriteRegister
debug	libnfc.driver.acr122_usb	TX: 6f 0d 00 00 00 00 00 00 00 00 ff 00 00 00 08 d4 08 63 03 8b 63 0d 00
debug	libnfc.driver.acr122_usb	RX: 80 08 00 00 00 00 00 00 81 00 d5 03 32 01 06 07 90 00
error	libnfc.driver.acr122_usb	Command Code verification failed
```

For some reason the `pcsc_scan` does not work on GNU Guix:
```
```
# /gnu/store/xymj77nibwqs1fvqblx7wjq48gpcjcz7-pcsc-tools-1.6.0/bin/pcsc_scan
Using reader plug'n play mechanism
Scanning present readers...
Waiting for the first reader... | SCardGetStatusChange: Command cancelled.
```

phodina, 2022-11-10:
Based on the debug info do you @Gnu-Hack think that the HW reader I have is different inside? I mean it responds differently than what the library expects.

My assumption is based on the TX is the byte array that's send over USB for configuration and RX is the respond from the device.

Gnu-Hack, 2022-11-10:
> Based on the debug info do you ***@***.***(https://github.com/Gnu-Hack) think that the HW reader I have is different inside? I mean it responds differently than what the library expects.

what is your reader?

José CardeLL

-------- Mensaje original --------

phodina, 2022-11-10:
Well it enumerates as:
```
$ lsusb
...
Bus 001 Device 013: ID 072f:2200 Advanced Card Systems, Ltd ACR122U
```

The text on the back of the NFC device says:
```
ACR122U
P/N: ACR122U-A9
```

phodina, 2022-11-20:
@Gnu-Hack any thoughts?

Cioci89, 2024-04-17:
I have the same problem with a new installation of Ubuntu Desktop 22.04.04

But the problem that i have found is in the libnfc6....

follow my guide for another problem...that resolve also this problem
https://github.com/nfc-tools/libnfc/issues/684#issuecomment-1632260381

Ferenc-, 2025-01-11:
Confirming that the workaround by @Cioci89 still works.