nfc-tools/libnfc #679: BUG in nfc_initiator_select_passive_target

versaloon, 2022-09-29:
`int
nfc_initiator_select_passive_target(nfc_device *pnd,
                                    const nfc_modulation nm,
                                    const uint8_t *pbtInitData, const size_t szInitData,
                                    nfc_target *pnt)
{
  uint8_t *abtInit = NULL;
  uint8_t maxAbt = MAX(12, szInitData);
  uint8_t *abtTmpInit = malloc(sizeof(uint8_t) * maxAbt);
  size_t  szInit = 0;
  int res;
  if ((res = nfc_device_validate_modulation(pnd, N_INITIATOR, &nm)) != NFC_SUCCESS) {
    free(abtTmpInit);
    return res;
  }
  if (szInitData == 0) {
    // Provide default values, if any
    prepare_initiator_data(nm, &abtInit, &szInit);
    free(abtTmpInit);
  } else if (nm.nmt == NMT_ISO14443A) {
    abtInit = abtTmpInit;
    iso14443_cascade_uid(pbtInitData, szInitData, abtInit, &szInit);
  } else {
    abtInit = abtTmpInit;
    memcpy(abtInit, pbtInitData, szInitData);
    free(abtTmpInit);
    szInit = szInitData;
  }
  HAL(initiator_select_passive_target, pnd, nm, abtInit, szInit, pnt);

  free(abtTmpInit);
}`

In this part:
`    abtInit = abtTmpInit;
    memcpy(abtInit, pbtInitData, szInitData);
    free(abtTmpInit);
    szInit = szInitData;`

abtInit is abtTmpInit, will be used after freed.

versaloon, 2022-09-29:
To avoid free abtInit used in initiator_select_passive_target, maybe use variable array will be better.
```c
int
nfc_initiator_select_passive_target(nfc_device *pnd,
                                    const nfc_modulation nm,
                                    const uint8_t *pbtInitData, const size_t szInitData,
                                    nfc_target *pnt)
{
  uint8_t *abtInit = NULL;
  uint8_t maxAbt = MAX(12, szInitData);
  uint8_t abtTmpInit[maxAbt];
  size_t  szInit = 0;
  int res;
  if ((res = nfc_device_validate_modulation(pnd, N_INITIATOR, &nm)) != NFC_SUCCESS) {
    return res;
  }
  if (szInitData == 0) {
    // Provide default values, if any
    prepare_initiator_data(nm, &abtInit, &szInit);
  } else if (nm.nmt == NMT_ISO14443A) {
    abtInit = abtTmpInit;
    iso14443_cascade_uid(pbtInitData, szInitData, abtInit, &szInit);
  } else {
    abtInit = abtTmpInit;
    memcpy(abtInit, pbtInitData, szInitData);
    szInit = szInitData;
  }
  HAL(initiator_select_passive_target, pnd, nm, abtInit, szInit, pnt);
}
```

tungnduong, 2022-12-30:
There's already a call to free(abtTmpInit) after the call to `Hal`.   It seems to me the better fix might be to remove the premature frees ? Something like:

`
{
  uint8_t *abtInit = NULL;
  uint8_t maxAbt = MAX(12, szInitData);
  uint8_t *abtTmpInit = malloc(sizeof(uint8_t) * maxAbt);
  size_t  szInit = 0;
  int res;
  if ((res = nfc_device_validate_modulation(pnd, N_INITIATOR, &nm)) != NFC_SUCCESS) {
    free(abtTmpInit); 
    return res;
  }
  if (szInitData == 0) {
    // Provide default values, if any
    prepare_initiator_data(nm, &abtInit, &szInit);
  } else if (nm.nmt == NMT_ISO14443A) {
    abtInit = abtTmpInit;
    iso14443_cascade_uid(pbtInitData, szInitData, abtInit, &szInit);
  } else {
    abtInit = abtTmpInit;
    memcpy(abtInit, pbtInitData, szInitData);
    szInit = szInitData;
  }
  HAL(initiator_select_passive_target, pnd, nm, abtInit, szInit, pnt);
  free(abtTmpInit);
}`

neomilium, 2023-10-21:
Could you provide a PR with patches? Thanks

closed 2023-10-21 by neomilium