nfc-tools/libnfc #684: nfc-mfclassic sucess but not wriring...

mvetois, 2023-01-13:
Hello, i have an issue with nfc-mfclassic. I try to copy a card to an other, the result of nfc-mfclassic is success, the data inside the nfc key is exactly the same as before the run of the command.

nfc-mfclassic output :
```
NFC reader: ACS / ACR122U PICC Interface opened
Found MIFARE Classic card:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04
       UID (NFCID1): 0c  68  13  3d
      SAK (SEL_RES): 08
RATS support: no
Guessing size: seems to be a 1024-byte card
Sent bits:     50  00  57  cd
Sent bits:     40 (7 bits)
Received bits: a (4 bits)
Sent bits:     43
Received bits: 0a
Card unlocked
Writing 64 blocks |................................................................|
Done, 64 of 64 blocks written.
```

I tried with an other nfc key, it's the same...

Some of you has any idea ?

Ludo1598, 2023-01-22:
Hello,
I have the same problem, did you find a solution to the problem ?

mvetois, 2023-01-25:
No, I haven't resolved this clue ..

Cioci89, 2023-07-12:
I have the same problem but i have a workaround..works for me..the problem there is only with libnfc-1.8.0....with 1.7.1 it works perfectly
This is the step:
**#Uninstall libnfc-1.8.0 that have the problem**
_sudo apt remove libnfc6_

I prepared a package of libnfc-1.7.1 with patched drivers for acr122 (i take the file acr122_usb.c of libnfc-1.8.0 and replaced this file libnfc-1.7.1/libnfc/drivers/acr122_usb.c)

**If you can download from here or from the attachment**
https://mega.nz/file/1l8gEAya#7FuZyhBjdn_U4oyYCoCn1QLvAdm4MorVkeTVwLjG6Fc

**#Now we need to unzip the file**
_unzip libnfc-1.7.1.zip
cd libnfc-1.7.1_
**#just for security that you are able to compile the package**
_sudo apt-get update
sudo apt-get install libnfc-dev build-essential libssl-dev
./configure
sudo make
sudo make install_

**#if you don't receive any error try to open a terminal and do:**
_nfc-list_
you need to see: nfc-list uses libnfc 1.7.1 (with the correct version)
if you see: nfc-list uses libnfc 1.8.0 (you need to remove the installed 1.8.0 version and after that you can reconfigure and make your old package but work for me)

**#If is all correct now you are able to write the file with nfc-mfclassic**
_nfc-mfclassic W a dumpfile.mfd_

i hope that this guide can help someone like me..

[libnfc-1.7.1.zip](https://github.com/nfc-tools/libnfc/files/12026826/libnfc-1.7.1.zip)

Ludo1598, 2023-07-13:
> I have the same problem but i have a workaround..works for me..the problem there is only with libnfc-1.8.0....with 1.7.1 it works perfectly This is the step: **#Uninstall libnfc-1.8.0 that have the [...]

Thank you very much for your solution, it works very well for me.
thanks again

ld21, 2023-07-14:
Unfortunately, my problem of trying to write to a card is not resolved with switching to 1.7.1. Even though I can easily read from cards, I cannot write to it:

`nfc-mfclassic W a dump.mfd` 
results in 
`NFC reader: pn532_uart:/dev/ttyUSB0 opened
Found MIFARE Classic card:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04  
       UID (NFCID1): d6  10  b6  26  
      SAK (SEL_RES): 08  
Guessing size: seems to be a 1024-byte card
Sent bits:     50  00  57  cd  
Sent bits:     40 (7 bits)
unlock failure!`

When trying to write it with
`nfc-mfclassic w a dump.mfd dump.mfd`
I get 
`NFC reader: pn532_uart:/dev/ttyUSB0 opened
Found MIFARE Classic card:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04  
       UID (NFCID1): d6  10  b6  26  
      SAK (SEL_RES): 08  
Guessing size: seems to be a 1024-byte card
Writing 64 blocks |x`

When trying with a highter LOG-Level and `LIBNFC_LOG_LEVEL=3  nfc-mfclassic w a dump.mfd dump.mfd`
I get
`debug	libnfc.config	key: [allow_autoscan], value: [true]
debug	libnfc.config	key: [allow_intrusive_scan], value: [true]
debug	libnfc.config	key: [device.name], value: [Adafruit PN532 Breakout]
debug	libnfc.config	key: [device.connstring], value: [pn532_uart:/dev/ttyUSB0]
debug	libnfc.config	key: [device.name], value: [Adafruit PN532 board via UART]
debug	libnfc.config	key: [device.connstring], value: [pn532_uart:/dev/ttyUSB0]
debug	libnfc.config	key: [device.allow_intrusive_scan], value: [true]
info	libnfc.config	Unknown key in config line: device.allow_intrusive_scan = true
debug	libnfc.config	key: [device.log_level], value: [3]
info	libnfc.config	Unknown key in config line: device.log_level = 3
debug	libnfc.config	key: [device.name], value: [PN532 board via UART]
debug	libnfc.config	key: [device.connstring], value: [pn532_uart:/dev/ttyUSB0]
debug	libnfc.config	key: [device.allow_intrusive_scan], value: [true]
info	libnfc.config	Unknown key in config line: device.allow_intrusive_scan = true
debug	libnfc.general	log_level is set to 3
debug	libnfc.general	allow_autoscan is set to true
debug	libnfc.general	allow_intrusive_scan is set to true
debug	libnfc.general	3 device(s) defined by user
debug	libnfc.general	  #0 name: "Adafruit PN532 Breakout", connstring: "pn532_uart:/dev/ttyUSB0"
debug	libnfc.general	  #1 name: "Adafruit PN532 board via UART", connstring: "pn532_uart:/dev/ttyUSB0"
debug	libnfc.general	  #2 name: "PN532 board via UART", connstring: "pn532_uart:/dev/ttyUSB0"
debug	libnfc.driver.pn532_uart	Attempt to open: /dev/ttyUSB0 at 115200 bauds.
debug	libnfc.bus.uart	Serial port speed requested to be set to 115200 bauds.
debug	libnfc.chip.pn53x	Diagnose
debug	libnfc.chip.pn53x	Timeout value: 500
debug	libnfc.bus.uart	TX: 55 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
debug	libnfc.chip.pn53x	SAMConfiguration
debug	libnfc.chip.pn53x	Timeout value: 1000
debug	libnfc.bus.uart	TX: 00 00 ff 03 fd d4 14 01 17 00 
debug	libnfc.bus.uart	RX: 00 00 ff 00 ff 00 
debug	libnfc.chip.pn53x	PN53x ACKed
debug	libnfc.bus.uart	RX: 00 00 ff 02 fe 
debug	libnfc.bus.uart	RX: d5 15 
debug	libnfc.bus.uart	RX: 16 00 
debug	libnfc.bus.uart	TX: 00 00 ff 09 f7 d4 00 00 6c 69 62 6e 66 63 be 00 
debug	libnfc.bus.uart	RX: 00 00 ff 00 ff 00 
debug	libnfc.chip.pn53x	PN53x ACKed
debug	libnfc.bus.uart	RX: 00 00 ff 09 f7 
debug	libnfc.bus.uart	RX: d5 01 
debug	libnfc.bus.uart	RX: 00 6c 69 62 6e 66 63 
debug	libnfc.bus.uart	RX: bc 00 
[... 146 lines cut ...]
debug	libnfc.bus.uart	TX: 00 00 ff 04 fc d4 32 01 00 f9 00 
debug	libnfc.bus.uart	RX: 00 00 ff 00 ff 00 
debug	libnfc.chip.pn53x	PN53x ACKed
debug	libnfc.bus.uart	RX: 00 00 ff 02 fe 
debug	libnfc.bus.uart	RX: d5 33 
debug	libnfc.bus.uart	RX: f8 00 
debug	libnfc.chip.pn53x	PowerDown
debug	libnfc.bus.uart	TX: 00 00 ff 03 fd d4 16 f0 26 00 
debug	libnfc.bus.uart	RX: 00 00 ff 00 ff 00 
debug	libnfc.chip.pn53x	PN53x ACKed
debug	libnfc.bus.uart	RX: 00 00 ff 03 fd 
debug	libnfc.bus.uart	RX: d5 17 
debug	libnfc.bus.uart	RX: 00 
debug	libnfc.bus.uart	RX: 14 00 
x
` 

Or when running `LIBNFC_LOG_LEVEL=3  nfc-mfclassic W a dump.mfd`
I get among similar code to above
`debug	libnfc.chip.pn53x	Chip error: "Timeout" (01), returned error: "RF Transmission Error" (-20))
unlock failure!
`

Any ideas what could be wrong?
Thanks!

Cioci89, 2023-07-16:
> Unfortunately, my problem of trying to write to a card is not resolved with switching to 1.7.1. Even though I can easily read from cards, I cannot write to it:
>
> `nfc-mfclassic W a dump.mfd`
> results [...]

I read "unlock failure!" and this is because you use nfc-mfclassic with W Uppercase without using a Chiese Magic Card with Sector 0 (uid) editable

ld21, 2023-07-16:
That's true, thank you Cioci89  – but look at the version where I use «w» (and I think I should be using w, because I'm simply trying to overwrite a card using a dump that I received from that very card).

In the log text after issuing `nfc-mfclassic w a dump.mfd dump.mfd` I receive a 
`Chip error: "CRC Error" (02), returned error: "RF Transmission Error" (-20))`. 

What could that be related to?

Ludo1598, 2023-07-16:
> That's true, thank you Cioci89 – but look at the version where I use «w» (and I think I should be using w, because I'm simply trying to overwrite a card using a dump that I received from that very [...]

Good morning
have you tried changing the name of the dumps
dump1 and dump2
`nfc-mfclassic w a dump1.mfd dump2.mfd`

ld21, 2023-07-17:
I have tried using two different files – it makes no difference for the writing-process.

elbowz, 2025-08-22:
Same issue:

Writing to a card appears to succeed, but the card content/keys are not actually updated.

```
$ nfc-mfclassic w a dest.dmp src.dmp 
NFC reader: PN532_UART opened
Found MIFARE Classic card:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04  
       UID (NFCID1): e9  b6  ee  b9  
      SAK (SEL_RES): 08  
Guessing size: seems to be a 1024-byte card
Writing 64 blocks |...............................................................|
Done, 63 of 64 blocks written.
```
However, the card remains unchanged — the keys/data are not updated.

Swapping the arguments (dest.dmp and src.dmp) or even writing the same dump twice produces the same result: no actual changes on the card.

Downgrading to version 1.7.1 fixes the issue, but that introduces dependency conflicts with other software.

Thanks!