nfc-tools/libnfc #690: Cannot detect Yubikey 5 NFC with PN532

agorgl, 2023-03-10:
I have a PN532 module plugged in HSU (UART) mode with a usb serial adapter.
Using `nfc-poll -v` and scanning various RFID cards and tags works flawlessly, but trying  to scan a `YubiKey 5 NFC` doesn't read anything.

I tried two different PN532 modules, one `ITEAD PN532 NFC MODULE` and an `ELECHOUSE NFC RFID MODULE V3` that exhibit same behavior.

Any idea on how to debug / diagnose the problem?

agorgl, 2023-03-15:
Scanning `yubikey` with NFC Tools on Android it shows up as having a tag type of `ISO 14443-4` with the description `NXP - Mifare Plus`. Same behavior is observed with another (train) nfc card that I have with tag type of `ISO 14443-4` and description `NXP - Mifare DESFire EV1 4k`.

Are these kinds of cards supported by `libnfc` / `nfc-poll`?
I also tried using another USB UART / TTL adapter, even serial over BT with a `HC-05` module, powering module with separate supply etc. just to be sure that it was not the problem of any other component in the equation.

gianklug, 2023-11-20:
Did you find out anything?

agorgl, 2023-11-22:
Nop, I kinda gave up after a while (I even tried to make my own NFC antenna!).
I think that these cheap PN532 modules are not powerful enough to interact with Yubikey NFC / Mifare DESFire EV1 4k cards

3rdIteration, 2023-11-25:
I have it detecting a Yubikey 5 NFC just fine, even via pcsc_scan. (Running on a Raspberry Pi 4 with PN532 connected via i2c)

libnfc + ifdnfc built from source using latest master of both repos.

agorgl, 2023-11-25:
I tried only using uart, can you try and tell us if it works with uart too? If so, what board do you have exactly?

3rdIteration, 2023-11-26:
Can't test UART at the moment, but it's this board here
https://a.aliexpress.com/_mqQRXte

It also detects the yubikey with plain old nfc-poll.

agorgl, 2023-11-26:
Are you sure this is your board? Because I have a similar chinese board that looks exactly the same, and I could not make yubikey scanning work with it in UART mode

3rdIteration, 2023-11-26:
That's the exact board that I am using

3rdIteration, 2023-12-06:
So I just checked and basically the quality varies dramatically between these cheap clone boards from AliExpress or Amazon... One of them ones that I have here has very poor range and struggles to even reliability detect full sized smart card, let alone a Yubikey...

agorgl, 2023-12-07:
Even from the same seller/vendor?