nfc-tools/libnfc #699: Error libnfc.driver.pn532_uart Application level error detected

NxFerrara, 2023-07-26:
I am trying to operate a PN532 from HiLetGo over UART connection (not usb) using libnfc on a Le Potato (AML-S905X-CC) with raspbian image "2022-09-22-raspbian-bullseye-arm64-lite+aml-s905x-cc.img"

When I do nfc-scan-device, it seems to correctly identify it:
nfc-scan-device
nfc-scan-device uses libnfc libnfc-1.8.0-65-g42de50f
1 NFC device(s) found:
- PN532 board via UART:
    pn532_uart:/dev/ttyAML6

But when I do nfc-list:
nfc-list
NFC device: PN532 board via UART opened
error   libnfc.driver.pn532_uart        Application level error detected
error   libnfc.driver.pn532_uart        Application level error detected

I am trying to troubleshoot this since when I try using another library "nfcpy" to detect the device, it fails:
python3 -m nfc --search-tty
This is the 1.0.4 version of nfcpy run in Python 3.9.2
on Linux-6.1.26-05272-g26c406245a2c-aarch64-with-glibc2.31
I'm now searching your system for contactless devices
Sorry, but I couldn't find any contactless device

Also I tried making a program to explicitly find the device on /dev/ttyAML6 which is where it is located.

Please let me know if there is anything else I should clarify.

Thank you!

NxFerrara, 2023-07-26:
I want to add on that the polling seems to work:
./nfc-poll
/src/libnfc/examples/.libs/nfc-poll uses libnfc libnfc-1.8.0-65-g42de50f
NFC reader: PN532 board via UART opened
NFC device will poll during 36000 ms (20 pollings of 300 ms for 6 modulations)
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04
       UID (NFCID1): 23  ff  94  95
      SAK (SEL_RES): 08
Waiting for card removing...nfc_initiator_target_is_present: Target Released
done.

F9Alejandro, 2023-08-09:
Is your user in the dialout group? It could possibly contribute to some issues. You could also try using sudo when running the commands to see if that works as well (if not already running under root)

closed 2023-08-09 by NxFerrara