nfc-tools/libnfc #708: serial speed for pn532

paweljasinski, 2024-01-15:
Hi,

are serial speed other than 115200 supported when using pn532?
Both:
<code>device.connstring = "pn532_uart:/dev/ttyUSB0:57600</code>
and
<code>device.connstring = "pn532_uart:/dev/ttyUSB0:460800</code>
produce a code dump.

--pawel

paweljasinski, 2024-03-11:
To be sure I am not misunderstanding anything. 
- establish communication with the default 115200,
- issue the SetSerialBaudRate
- re-establish communication with a new speed

NFC-funs, 2024-03-13:
Please refer to the diagram:
[image]
To be sure that the baud rate-"Real HSU Freq" error is not greater than 4%:
[image]

[PN532 User Manual V0.2.pdf](https://github.com/nfc-tools/libnfc/files/14584891/PN532.User.Manual.V0.2.pdf)
[PN532_C1.pdf](https://github.com/nfc-tools/libnfc/files/14584892/PN532_C1.pdf)

paweljasinski, 2024-03-13:
already have a basic communication with 230.4.

> To be sure that the baud rate-"Real HSU Freq" error is not greater than 4%:
Do I have to adjust registers value (HSU_CNT and HSU_PRE) myself, or it happens as part of SetSerialBaudRate?

I have another question:
Once the comm speed is changed from default, it is persistent until reset or power cycle.
For now I am codding with assumption that libnfc works with the board after reset and starts with 115200 and changes to requested speed. The code is not trying to reestablished communication after program terminated but board did not get reset and is still using the altered speed.
Is this something you would considered to acceptable in PR?