nfc-tools/libnfc #749: CMake problem

nazaroaversano, 2024-12-07:
[  1%] Building C object libnfc/CMakeFiles/nfc.dir/__/contrib/win32/stdlib.c.obj
In file included from C:\dev\libnfc-1.8.0\contrib\win32\stdlib.c:35:
C:/dev/libnfc-1.8.0/contrib/windows.h:47:13: warning: "ETIMEDOUT" redefined
   47 | #    define ETIMEDOUT     WSAETIMEDOUT
      |             ^~~~~~~~~
In file included from C:/mingw64/lib/gcc/x86_64-w64-mingw32/14.2.0/include/mm_malloc.h:29,
                 from C:/mingw64/lib/gcc/x86_64-w64-mingw32/14.2.0/include/xmmintrin.h:34,
                 from C:/mingw64/lib/gcc/x86_64-w64-mingw32/14.2.0/include/emmintrin.h:31,
                 from C:/mingw64/x86_64-w64-mingw32/include/winnt.h:2430,
                 from C:/mingw64/x86_64-w64-mingw32/include/minwindef.h:163,
                 from C:/mingw64/x86_64-w64-mingw32/include/windef.h:9,
                 from C:/mingw64/x86_64-w64-mingw32/include/windows.h:69,
                 from C:/dev/libnfc-1.8.0/contrib/windows.h:36:
C:/mingw64/x86_64-w64-mingw32/include/errno.h:223:9: note: this is the location of the previous definition
  223 | #define ETIMEDOUT 138
      |         ^~~~~~~~~
C:/dev/libnfc-1.8.0/contrib/windows.h:48:13: warning: "ENOTSUP" redefined
   48 | #    define ENOTSUP       WSAEOPNOTSUPP
      |             ^~~~~~~
C:/mingw64/x86_64-w64-mingw32/include/errno.h:76:9: note: this is the location of the previous definition
   76 | #define ENOTSUP         129
      |         ^~~~~~~
C:/dev/libnfc-1.8.0/contrib/windows.h:49:13: warning: "ECONNABORTED" redefined
   49 | #    define ECONNABORTED  WSAECONNABORTED
      |             ^~~~~~~~~~~~
C:/mingw64/x86_64-w64-mingw32/include/errno.h:102:9: note: this is the location of the previous definition
  102 | #define ECONNABORTED 106
      |         ^~~~~~~~~~~~
C:\dev\libnfc-1.8.0\contrib\win32\stdlib.c: In function 'setenv':
C:\dev\libnfc-1.8.0\contrib\win32\stdlib.c:43:12: error: passing argument 1 of 'strcpy' from incompatible pointer type [-Wincompatible-pointer-types]
   43 |     strcpy(str, name);
      |            ^~~
      |            |
      |            char **
In file included from C:/mingw64/x86_64-w64-mingw32/include/guiddef.h:154,
                 from C:/mingw64/x86_64-w64-mingw32/include/winnt.h:648:
C:/mingw64/x86_64-w64-mingw32/include/string.h:61:45: note: expected 'char * restrict' but argument is of type 'char **'
   61 |   char * __cdecl strcpy(char * __restrict__ _Dest,const char * __restrict__ _Source);
      |                         ~~~~~~~~~~~~~~~~~~~~^~~~~
C:\dev\libnfc-1.8.0\contrib\win32\stdlib.c:44:12: error: passing argument 1 of 'strcat' from incompatible pointer type [-Wincompatible-pointer-types]
   44 |     strcat(str, "=");
      |            ^~~
      |            |
      |            char **
C:/mingw64/x86_64-w64-mingw32/include/string.h:62:45: note: expected 'char * restrict' but argument is of type 'char **'
   62 |   char * __cdecl strcat(char * __restrict__ _Dest,const char * __restrict__ _Source);
      |                         ~~~~~~~~~~~~~~~~~~~~^~~~~
C:\dev\libnfc-1.8.0\contrib\win32\stdlib.c:45:12: error: passing argument 1 of 'strcat' from incompatible pointer type [-Wincompatible-pointer-types]
   45 |     strcat(str, value);
      |            ^~~
      |            |
      |            char **
C:/mingw64/x86_64-w64-mingw32/include/string.h:62:45: note: expected 'char * restrict' but argument is of type 'char **'
   62 |   char * __cdecl strcat(char * __restrict__ _Dest,const char * __restrict__ _Source);
      |                         ~~~~~~~~~~~~~~~~~~~~^~~~~
C:\dev\libnfc-1.8.0\contrib\win32\stdlib.c:46:19: error: passing argument 1 of 'putenv' from incompatible pointer type [-Wincompatible-pointer-types]
   46 |     return putenv(str);
      |                   ^~~
      |                   |
      |                   char **
In file included from C:/mingw64/lib/gcc/x86_64-w64-mingw32/14.2.0/include/mm_malloc.h:27:
C:/mingw64/x86_64-w64-mingw32/include/stdlib.h:715:34: note: expected 'const char *' but argument is of type 'char **'
  715 |   int __cdecl putenv(const char *_EnvString) __MINGW_ATTRIB_DEPRECATED_MSVC2005;
      |                      ~~~~~~~~~~~~^~~~~~~~~~
C:\dev\libnfc-1.8.0\contrib\win32\stdlib.c: In function 'unsetenv':
C:\dev\libnfc-1.8.0\contrib\win32\stdlib.c:54:10: error: passing argument 1 of 'strcpy' from incompatible pointer type [-Wincompatible-pointer-types]
   54 |   strcpy(str, name);
      |          ^~~
      |          |
      |          char **
C:/mingw64/x86_64-w64-mingw32/include/string.h:61:45: note: expected 'char * restrict' but argument is of type 'char **'
   61 |   char * __cdecl strcpy(char * __restrict__ _Dest,const char * __restrict__ _Source);
      |                         ~~~~~~~~~~~~~~~~~~~~^~~~~
C:\dev\libnfc-1.8.0\contrib\win32\stdlib.c:55:10: error: passing argument 1 of 'strcat' from incompatible pointer type [-Wincompatible-pointer-types]
   55 |   strcat(str, "=");
      |          ^~~
      |          |
      |          char **
C:/mingw64/x86_64-w64-mingw32/include/string.h:62:45: note: expected 'char * restrict' but argument is of type 'char **'
   62 |   char * __cdecl strcat(char * __restrict__ _Dest,const char * __restrict__ _Source);
      |                         ~~~~~~~~~~~~~~~~~~~~^~~~~
C:\dev\libnfc-1.8.0\contrib\win32\stdlib.c:56:10: error: passing argument 1 of 'putenv' from incompatible pointer type [-Wincompatible-pointer-types]
   56 |   putenv(str);
      |          ^~~
      |          |
      |          char **
C:/mingw64/x86_64-w64-mingw32/include/stdlib.h:715:34: note: expected 'const char *' but argument is of type 'char **'
  715 |   int __cdecl putenv(const char *_EnvString) __MINGW_ATTRIB_DEPRECATED_MSVC2005;
      |                      ~~~~~~~~~~~~^~~~~~~~~~
mingw32-make[2]: *** [libnfc\CMakeFiles\nfc.dir\build.make:319: libnfc/CMakeFiles/nfc.dir/__/contrib/win32/stdlib.c.obj] Error 1
mingw32-make[1]: *** [CMakeFiles\Makefile2:320: libnfc/CMakeFiles/nfc.dir/all] Error 2
mingw32-make: *** [Makefile:155: all] Error 2

nazaroaversano, 2024-12-07:
I can't compile...

mikma, 2025-05-11:
The  `str` declarations in contrib/win32/stdlib.c (lines 42 and 53 below) looks odd. It shouldn't be an array of char pointers (`char *str[X]`), but an array of char (`char str[X]`). Maybe the arrays also need to be larger since changing from pointers to char means the allocated size will shrink with 87.5% on 64 bit architectures, and there doesn't seem to be any check for buffer overwrites.

https://github.com/nfc-tools/libnfc/blob/3fa0751ad58fb0053d3de2e61bbbd4066259104c/contrib/win32/stdlib.c#L37-L57

fuuhoo, 2025-06-25:
> I can't compile...

Has the problem been resolved?

fuuhoo, 2025-06-26:
> > I can't compile...
> 
> Has the problem been resolved?

Modifying to the following code is okay
```
int setenv(const char *name, const char *value, int overwrite)
{
  char *env = getenv(name);
  if ((env && overwrite) || (!env)) {
    char str[32];
    strcpy(str, name);
    strcat(str, "=");
    strcat(str, value);
    return putenv(str);
  }
  return -1;
}

void unsetenv(const char *name)
{
  char str[32];
  strcpy(str, name);
  strcat(str, "=");
  putenv(str);
}

```